JAndrassy/TelnetStream #5: Adding to Arduino Library Manager

ameer1234567890, 2020-05-22:
Considering that this library has been existent for quite some time now and is pretty stable, I suggest that this be added to Arduino's library manager. In order to do that a tag needs to be created with the version number in `library.properties` and an issue needs to be opened at https://github.com/arduino/Arduino/issues with title `[Library Manager] TelnetStream Library` and the URL to this library in the issue body.

jandrassy, 2020-05-22:
I have [4 serious libraries](https://www.arduinolibraries.info/authors/jandrassy) in Library Manager. I know how to add it. To have this one in library manager I would have to write a documentation and add examples.
btw right know I try to make it work with the Ethernet library.

jandrassy, 2020-07-25:
done

closed 2020-07-25 by jandrassy