JAndrassy/TelnetStream #7: Error compiling TelnetStreamEthTest with EthernetENC

jandrassy, 2020-10-03:
gcc linker fails for EthernetENC library on all architectures with "EthernetServer.cpp.o: plugin needed to handle lto object"
```
arduino-1.8.13/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: EthernetServer.cpp.o: plugin needed to handle lto object
```
this is some error of gcc with dot-a-linked library

the TelnetPrintEthTest verifies without problems.

possible workarounds:
- in EthernetENC in library.properties remove `dot_a_linkage=true`. consequence is a larger build result
- use TelnetPrint

Staggeble, 2021-02-08:
I have issues while compiling the TelnetStream.h library with the latest Arduino 1.8.13.

error: missing binary operator before token "("
 #if __has_include(<EthernetENC.h>)

Is this issue linked with the above?

jandrassy, 2021-02-09:
> I have issues while compiling the TelnetStream.h library with the latest Arduino IDE 1.8.13.
> 
> error: missing binary operator before token "("
> #if __has_include(<EthernetENC.h>)
> 
> Is this issue linked with the above?

the version of IDE doesn't matter. what MCU is it?

Staggeble, 2021-02-09:
> > I have issues while compiling the TelnetStream.h library with the latest Arduino IDE 1.8.13.
> > error: missing binary operator before token "("
> > #if __has_include(<EthernetENC.h>)
> > Is this issue linked with the above?
> 
> the version of IDE doesn't matter. what MCU is it?

I am using an ESP-01S.

jandrassy, 2021-02-09:
> I am using an ESP-01S.

this issue is about a different problem.
the example for esp8266 compiles without problems. 
what version of esp8266 arduino do you use?
did you change something in the library?

Staggeble, 2021-02-09:
> > I am using an ESP-01S.
> 
> this issue is about a different problem.
> the example for esp8266 compiles without problems.
> what version of esp8266 arduino do you use?
> did you change something in the library?

Yesterday I pulled the latest master branch from GitHub.
https://github.com/esp8266/Arduino.git
I didn't modify anything. I will try if the example also works here.

jandrassy, 2021-02-09:
> Yesterday I pulled the latest master branch from GitHub.
> https://github.com/esp8266/Arduino.git
> I didn't modify anything. I will try if the example also works here.

try fixed NetTypes.h

Staggeble, 2021-02-10:
I believe the problem was at my side. The actual esp8266 Arduino library installed was version 2.3. Yesterday I updated it to 2.7.4 and it worked without any problem. I tried to install it via the .zip file, but I had to install it via the board manager.

svdrummer, 2021-04-16:
Can we assume this issue is now closed? Seems the issue with the ESP board manager update resolved the issue.

jandrassy, 2021-04-17:
> Can we assume this issue is now closed? Seems the issue with the ESP board manager update resolved the issue.

this is issue is not about esp

svdrummer, 2021-04-17:
> > Can we assume this issue is now closed? Seems the issue with the ESP board manager update resolved the issue.
> 
> this is issue is not about esp

"I believe the problem was at my side. The actual esp8266 Arduino library installed was version 2.3. Yesterday I updated it to 2.7.4 and it worked without any problem. I tried to install it via the .zip file, but I had to install it via the board manager."

jandrassy, 2021-04-17:
> "I believe the problem was at my side. The actual esp8266 Arduino library installed was version 2.3. Yesterday I updated it to 2.7.4 and it worked without any problem. I tried to install it via the [...]

you still didn't noticed that Staggeble hijacked the issue?

closed 2022-01-17 by jandrassy