JAndrassy/TelnetStream #8: ESP32 example - Need to .read() in main loop before .println("")

Wistrel, 2020-10-13:
In the ESP32 example "TelentStreamEsp32Test", if I replace the loop with:

void loop() 
{
  //TelnetStream.read();
     TelnetStream.println("this won't work");
     delay(100);
}

I see nothing when I telnet in.  However if I then un-comment the 1st line, it works.

Also see example here (line 14):

https://github.com/Wistrel/ESP32-OTA/blob/master/0TA_Template_Sketch_TelnetStream/0TA_Template_Sketch_TelnetStream.ino

jandrassy, 2020-10-13:
it is a bug in esp32.
https://github.com/espressif/arduino-esp32/issues/4399

RobAxt, 2020-10-14:
I have the same problem with an ESP8266 and found the behavior differs from version. In version 1.2.0 I have to use the TelnetStream.read() in order to work properly, but with the version 1.0.0 there is no need for that workaround. The version of the library used are the same.

|-- <TelnetStream> 1.0.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0

-- <TelnetStream> 1.2.0+sha.da6e2fc
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0

Thank you very much for this project!!

jandrassy, 2020-10-14:
> 
> 
> I have the same problem with an ESP8266 and found the behavior differs from version. In version 1.2.0 I have to use the TelnetStream.read() in order to work properly, but with the version 1.0.0 there is no need for that workaround. The version of the library used are the same.
> 
> |-- 1.0.0
> | |-- 1.0
> | |-- <Ethernet(esp8266)> 1.0.4
> | | |-- 1.0
> 
> -- 1.2.0+sha.da6e2fc
> | |-- 1.0
> | |-- <Ethernet(esp8266)> 1.0.4
> | | |-- 1.0
> 
> Thank you very much for this project!!

jandrassy, 2020-10-14:
sorry, it was something I didn't think of it.
it is fixed now. in a few hours it should be available in Library manager as 1.2.1

closed 2020-10-14 by jandrassy

Wistrel, 2020-10-15:
Thanks so much for the super fast fix! Really incredible work! Just tested and confirm this is working. 

Weirdly, I did manage to get a disconnect (and was able to repeat it multiple times), when running something that only outputs values, then holding the enter key down for a 5 seconds, releasing, repeating, maybe 4 or 5 times and then it would drop the connection. 

I had other stuff going on behind the scenes though so should really set up a more basic program to test. I'll come back to it and  raise as separate issue if I'm able to replicate in a more reliable/demonstrable way. This said though, this one might relate to the aforementioned HTTP server issue.

jandrassy, 2020-10-16:
yes, there was some days ago a discussion on esp8266 gitter about what happens if received dat are not read. it doesn't do good. 
the TelnetStream could have an option to discard any received data.