JAndrassy/TelnetStream #9: Number of clients

svdrummer, 2020-11-02:
I am using telnestream to send data to mutiple clients. Ie, serial data in, gets sent to telnetstream.

Now and then, a client drops off line and leaves a port open. after some time, i run out of ports.
I use CPU.reset() every hour just to clear the ports back to zero.

Is there a way of checking how many clients are connected?
Using TCP server i can, but i am not using TCP server.
PS, yes, I have looked on google before posting here.
Thank you

jandrassy, 2020-11-02:
on what hardware?

svdrummer, 2020-11-02:
ESP8266 NodeMCU and esp32.
ESP8266wifi etc
Thanks.

jandrassy, 2020-11-02:
esp8266 arduino has limited count of TCP connections at time. it is 4 or 5

svdrummer, 2020-11-02:
Wifi has 5 clients max by default. I am just looking for a way to see how many clients are connected.

jandrassy, 2020-11-02:
> Wifi has 5 clients max by default. I am just looking for a way to see how many clients are connected.

the clients are handled with ArduinoWiFiServer in this library so it is easy to add a function to count the active clients.

svdrummer, 2020-11-02:
Thank you 😁
I will search the internet for an example.

jandrassy, 2020-11-03:
the class is in this library and is a month old. you will not find anything about it with google.
my recommendation is, copy the ArduinoWiFiServer class from this library to your sketch and use it instead of theTelnetStream. you can then modify it as you need to count connected clients, abort connections etc

closed 2020-11-03 by jandrassy