JAndrassy/TelnetStream #10: Exception error

svdrummer, 2021-01-07:
Hello,
I have been using this library for months without an issue....however... i can simulate a crash on demand.
ESP8266 wemos D1 mini.
SoftAP (Soft access point only)
serial data comes on on Software serial port and is sent out ot telnet stream. Any client that connects can see the telnetstream.
This all works fine for months.
To cause a crash, i connect to the wemos server let the stream start then disconnect from the server. On my windows laptop its just connect and disconnect. On disconnect, the system crashes. If i run the code with telnetstream commented out and replaced with a print statement, no crashes occur.
I have tried TelnetStream2 and dont get the crashes, however, TelnetStream2 only supports one client at a time.
Regards and thanks

jandrassy, 2021-01-07:
you disconnect from AP of esp?
what is TelnetStream2?

svdrummer, 2021-01-07:
I have an esp8266 in access point mode.
I connect to that access point with a telnet client.
Whilst connected via telnet, if i turn off or switch the wifi connection, the esp crashes.
Telnetstream2 is on github. When using a text editor with compare function, it is similar by about 90% to this TelnetStream

jandrassy, 2021-01-07:
ok. I eliminated the crash. try master version from GitHub. 

the connection stays open until some very long TCP timeout I guess. If the computer connects back to SoftAP before the TCP timeout the telnet client continues receiving.

svdrummer, 2021-01-07:
Firstly, thank you for your response.
I simulated the the last issue and it has been resolved. At one point i had a connection but ni data streaming, however, i was connecting and disconnecting fast.
I have placed TelnetStream into a local folder, ie src/telnetstream/src/telnetstream.h. As Arduino sometimes uses other libraries.
Thanks you once again.
PS, your library is being used by me as i have a health problem and it is helping(all responsibility accepted by myself)

closed 2021-01-08 by jandrassy