JAndrassy/TelnetStream #12: TelnetStream Esp32 Ap mode

maramb81, 2021-03-26:
Hi

I'm looking for some example of Telnet Terminal with an ESP32 in AP mode, when I connected my pc to the esp 32 wifi connection, I use putty with IP 192.168.4.1

but it didn't work

could some one help me ?

Thank you

jandrassy, 2021-03-27:
with TelnetStreamEsp32Test example in esp32?

maramb81, 2021-03-27:
thank you for reply

I have already tried your solution, but works only if I use ESP 32 like a Client wife, so I need a router.
My goal is:

WiFi.softAP(ssid, pass);
TelnetStream.begin();

but that does not work

regards

jandrassy, 2021-03-27:
I am sure it works with SoftAP of esp32. There is nothing STA or SoftAP specific.

maramb81, 2021-03-27:
#include <WiFi.h>
#include <TelnetStream.h>

const char* ssid = "mySsidTest";  //  your network SSID (name)
const char* pass = "myPasswordTest";   // your network password

#define LED  2

void setup() {
  pinMode(LED, OUTPUT);  
  digitalWrite(LED, HIGH);
  
  Serial.begin(115200);
  Serial.println("setup");

  WiFi.softAP(ssid, pass);
  TelnetStream.begin();
  Serial.println("setup complete");
  
}

void loop() {

   static unsigned long next;
  if (millis() - next > 5000) {
    next = millis();
    log();
    Serial.println("log");
  }

}

void log() {
  static int i = 0;

  TelnetStream.print(i++);
  TelnetStream.print(" ");
  TelnetStream.print(" A0: ");
  TelnetStream.println(analogRead(A0));
}

*********************************
MBP:~$ telnet 192.168.4.1
Trying 192.168.4.1...
Connected to 192.168.4.1.
Escape character is '^]'.

jandrassy, 2021-03-27:
add TelnetStream.read() to loop.
https://github.com/jandrassy/TelnetStream/issues/8

closed 2021-03-27 by jandrassy

maramb81, 2021-03-27:
thank you so much, it works

TOP