JAndrassy/TelnetStream #13: TelnetStream doesn't stop on ESP32

Layochi, 2021-05-01:
Hello everyone!
I have a problem with this library... When using 
`TelnetStream.flush();`
`TelnetStream.stop();`
ESP32 does not close the communication.

I attach my code here so you can help me. The idea of this code is to close the communication when 'C' is written in the terminal. It works because the message "Closing the Telnet Communication..." is displayed but the communication remains effective. 

Thanks in advance for your help !

```
#include <TelnetStream.h>

void setupOTA() {
  Serial.println("Starting Telnet Communication...");
  TelnetStream.begin();
  Serial.println("Telnet Initialized!");

  Serial.println("You can open a new terminal and write the command:\n$ telnet " + WiFi.localIP().toString());
}

void debug(const char* string) {
  Serial.print(string);
  TelnetStream.print(String(string));
}

void debugln(const char* string) {
  Serial.println(string);
  TelnetStream.println(String(string));
}

void handle() {
  switch (TelnetStream.read()) {
    case 'C':
      TelnetStream.println("\nClosing the Telnet Communication...");
      TelnetStream.flush();
      TelnetStream.stop();
      break;
    case 'R':
      debugln("\nRebooting...");
      ESP.restart();
      break;
  }
}
```

jandrassy, 2021-05-01:
```
Trying 192.168.1.116...
Connected to 192.168.1.116.
Escape character is '^]'.
C

Closing the Telnet Communication...
Connection closed by foreign host.
```
ESP32 Arduino 1.0.6

Layochi, 2021-05-01:
Ok, I don't know what's going on my PC...

closed 2022-01-16 by jandrassy