JAndrassy/TelnetStream #14: Latest update to ESP8266 core code causes TelnetStream to silently fail

PuceBaboon, 2021-06-07:
Hi Juraj,

   First, many thanks for TelnetStream.  It turned out to be just what I wanted to implement a simple digital clock display ([](https://github.com/PuceBaboon/ESP-DigiClock)).  Unfortunately, it seems that a recent ESP8266 core code update is causing us problems.  My code suddenly started giving compile errors (non-fatal) and telnet stopped working.
~~~
In file included from .pio/libdeps/esp01_1m/TelnetStream/src/NetTypes.h:30,
                 from .pio/libdeps/esp01_1m/TelnetStream/src/TelnetStream.h:22,
                 from /home/PlatformIO/ESP01S_DS3231/src/alarm.ino:46:
/home/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src/Ethernet.h:39: warning: "MAX_SOCK_NUM" redefined
   39 | #define MAX_SOCK_NUM 8
      | 
In file included from /home/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:28,
                 from /home/PlatformIO/ESP01S_DS3231/src/alarm.ino:42:
/home/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/wl_definitions.h:42: note: this is the location of the previous definition
   42 | #define MAX_SOCK_NUM  4
      | 
In file included from /home/.platformio/packages/framework-arduinoespressif8266/libraries/ArduinoOTA/ArduinoOTA.h:5,
                 from /home/PlatformIO/ESP01S_DS3231/src/alarm.ino:47:
/home/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiUdp.h:28: warning: "UDP_TX_PACKET_MAX_SIZE" redefined
   28 | #define UDP_TX_PACKET_MAX_SIZE 8192
      | 
In file included from .pio/libdeps/esp01_1m/TelnetStream/src/NetTypes.h:30,
                 from .pio/libdeps/esp01_1m/TelnetStream/src/TelnetStream.h:22,
                 from /home/PlatformIO/ESP01S_DS3231/src/alarm.ino:46:
/home/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src/Ethernet.h:150: note: this is the location of the previous definition
  150 | #define UDP_TX_PACKET_MAX_SIZE 24
~~~
It seems like the __has_include(Ethernet.h) is causing the problem, so to fix my specific issue, I simply commented-out the whole section:-
~~~
// #elif __has_include(<Ethernet.h>)
// #include <Ethernet.h>
// #define NetClient EthernetClient
// #define NetServer EthernetServer
~~~
   This gets us back to a clean compile and a working application.

   Thought you might like to know, just in case anyone else sees the same thing.

   Best wishes from Japan,

          -John-

jandrassy, 2021-06-07:
I fixed it in February. the fix is not in a release yet, but here in the repository

closed 2021-06-07 by jandrassy

PuceBaboon, 2021-06-07:
Many thanks, Juraj.   I completely missed the key NetTtypes comment;
sorry!

                           -John-