JAndrassy/TelnetStream #17: Connection Refused

camsaway, 2021-11-03:
Hi,

I deployed this for the first time about 9 months ago on a couple of IOT sensors around my house. It initially worked fine so I suspect my code but don't know how to diagnose.

I constantly get "Connection Refused" from all my sensors. OTA still works so I don't think it's routing or firewall and I've tried a couple of different telnet clients and devices - all with the same problem.

Can you advise: common mistakes that could cause this (and if possible how to fix) error, and/or how to get more information other than Connection Refused (i.e. a connection refused reason).

This is all on ESP8266 (1 NodeMCU and a couple of D1 Minis).

Thanks in advance

jandrassy, 2021-11-03:
Hello. which version of the library and esp8266 core? and do you use TelnetStream or TelnetPrint object?

camsaway, 2021-11-04:
Hi,

Using TelnetStream Object, v1.2.1 and Espressif 8266 v3.2.0.

I doubt it makes a difference but I manage my code in Visual Studio Code using Platform.io plugin. So the versions of code are pulled automatically from their repo's.

The error is an immediate hard error, not a timeout, so something is actively sending back connection refused.

Thanks
Cameron.

jandrassy, 2021-11-04:
do the sensors connect at once? the count of concurrent TCP connections is limited on esp8266

camsaway, 2021-11-04:
Hi,

Each sensor is on a separate ESP so there should only be a connection every
10mins to AWS to send data for a second or less. There should be no other
connections and OTA and TelnetStream should be the only listening sockets.

Sent from my mobile phone so may be briefer than normal...

Regards

Cameron Gray
079 528 4802
***@***.***

https://www.linkedin.com/in/camerongray76

camsaway, 2021-11-06:
Have just run up a test NodeMCU with nothing but the example code and get the same problem. Will try going back a couple of versions for the Espressif platform and see

camsaway, 2021-11-06:
OK. v3.1.0 fails but v2.6.3 works - so seems something has changed here. I'll run my boards on 2.6.3 until there is a fix.

jandrassy, 2021-11-06:
so the latest version 3 of esp8266 core is 3.0.2 and the latest stable 2 is 2.7.4. why would you run 3.0.1 or 2.6?

camsaway, 2021-11-06:
Hi, Maybe we are talking different things here. It's the below platform versions I'm referring to.
They contain Arduino core 2.7.4 and 3.0.1 respectively.

https://github.com/platformio/platform-espressif8266/releases

jandrassy, 2021-11-06:
I see 3.2.0 with 3.0.2 there

camsaway, 2021-11-06:
That was the original version of was using, I tried the two older versions to see if worked as I went back

jandrassy, 2021-11-09:
sorry. the problem is with TelnetStream library and I fixed it a while ago,  but it is not in a release. please try the git master version

camsaway, 2021-11-10:
Ok. Sure. Thanks.

Regards

Cameron Gray
079 528 4802
***@***.***

https://www.linkedin.com/in/camerongray76

Aasim-A, 2022-01-16:
Hi @jandrassy, could you make a new release to resolve this issue because it took me a long time to figure it out.

jandrassy, 2022-01-16:
> Hi @jandrassy, could you make a new release to resolve this issue because it took me a long time to figure it out.

done. It will be in Library Manager in a few hours

closed 2022-01-16 by jandrassy