JAndrassy/TelnetStream #23: TelnetStream.stop();

ryaske, 2022-11-03:
TelnetStream.stop(); doesn't stop. I expect the connection to be dropped/disconnected on the client side, however nothing happens and instead I am still connected and receiving telnet messages.

JAndrassy, 2022-11-03:
multiple clients can connect to TelnetStream server. stop() should close first of them.

ryaske, 2022-11-04:
Thank you for the reply (and for the code)!

I am only connecting to the server once.

> multiple clients can connect to TelnetStream server. stop() should close first of them.

If this was the case, calling .stop() on a loop 100x would eventually close all of the connections. However, that is still not the case for me. stop() seems to have no effect what so ever, since I can continue to call TelnetStream.println() with success even after stop()!

Maybe it is a setting on my system side, because I see that the TCP connection is kept open and my telnet client does not know even when i remove power from the ESP. It still does not disconnect until the TCP stream times out. Which is exactly why I am looking for a way to gracefully end the connection from the TelnetStream server side.

Let me know if there is any additional troubleshooting I can do to help hash this out. I dont immediately suspect it is a problem with your library but I do know it is a problem for me

JAndrassy, 2024-02-05:
in esp32 WiFi library stop() does nothing. so the TelnetStream library can't stop the connection

closed 2024-02-05 by JAndrassy