JAndrassy/TelnetStream #24: Setting two telnet channels

svdrummer, 2022-12-17:
Is it possible to declater two instances of TelnetStream?

I am sending data to several units and wanted a different channel for speciffic data, rather than using the serial port.
Esp32 serial port resets the micro each time you connecte a terminal to the port. 

Ie 
TelnetStream Telnet1(23)
Telnet Stream Telnet2(100)

JAndrassy, 2022-12-17:
yes

svdrummer, 2022-12-17:
i can't seem to get it to compile. Must be the way i am configuring it.
As per the examples, i dont declare an object. 
I currently just do the following
TelnetStream.write(someMessage);
How would I declare two streams.? I have tried
TelnetStream Telnet1(23)
Telnet Stream Telnet2(100)

JAndrassy, 2022-12-17:
`TelnetStreamClass telnet(100);`

closed 2022-12-17 by svdrummer