JAndrassy/TelnetStream #26: error with es8266 with basic example

chepecarlos, 2023-01-21:
It does not compile with esp8266 in arduino ide 1.8.19

--- 
Arduino:1.8.19 (Linux), Tarjeta:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /home/chepecarlos/Arduino/libraries/TelnetStream/src/NetTypes.h:28,
                 from /home/chepecarlos/Arduino/libraries/TelnetStream/src/TelnetStream.h:22,
                 from /tmp/arduino_modified_sketch_653462/TelnetStreamEsp32Test.ino:3:
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h: In constructor 'ArduinoWiFiServer::ArduinoWiFiServer(const IPAddress&, uint16_t)':
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h:37:82: error: no matching function for call to 'WiFiServer::WiFiServer(const IPAddress&, uint16_t&)'
   37 |   ArduinoWiFiServer(const IPAddress& addr, uint16_t port) : WiFiServer(addr, port) {}
      |                                                                                  ^
In file included from /home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFi.h:32,
                 from /tmp/arduino_modified_sketch_653462/TelnetStreamEsp32Test.ino:1:
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:36:3: note: candidate: 'WiFiServer::WiFiServer(uint16_t)'
   36 |   WiFiServer(uint16_t);
      |   ^~~~~~~~~~
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:36:3: note:   candidate expects 1 argument, 2 provided
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:31:7: note: candidate: 'constexpr WiFiServer::WiFiServer(const WiFiServer&)'
   31 | class WiFiServer : public Server {
      |       ^~~~~~~~~~
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:31:7: note:   candidate expects 1 argument, 2 provided
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:31:7: note: candidate: 'constexpr WiFiServer::WiFiServer(WiFiServer&&)'
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFiServer.h:31:7: note:   candidate expects 1 argument, 2 provided
In file included from /home/chepecarlos/Arduino/libraries/TelnetStream/src/NetTypes.h:28,
                 from /home/chepecarlos/Arduino/libraries/TelnetStream/src/TelnetStream.h:22,
                 from /tmp/arduino_modified_sketch_653462/TelnetStreamEsp32Test.ino:3:
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h: In member function 'virtual size_t ArduinoWiFiServer::write(const uint8_t*, size_t)':
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h:77:32: error: 'ESTABLISHED' was not declared in this scope
   77 |         if (client.status() == ESTABLISHED) {
      |                                ^~~~~~~~~~~
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h:90:32: error: 'ESTABLISHED' was not declared in this scope
   90 |         if (client.status() == ESTABLISHED && client.availableForWrite() > 0) {
      |                                ^~~~~~~~~~~
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h: In member function 'ArduinoWiFiServer::operator bool()':
/home/chepecarlos/Arduino/libraries/TelnetStream/src/ArduinoWiFiServer.h:123:40: error: 'LISTEN' was not declared in this scope
  123 |   operator bool() {return (status() == LISTEN);}
      |                                        ^~~~~~
/tmp/arduino_modified_sketch_653462/TelnetStreamEsp32Test.ino: In function 'void setup()':
TelnetStreamEsp32Test:18:14: error: invalid conversion from 'const char*' to 'char*' [-fpermissive]
   18 |   WiFi.begin(ssid, pass);
      |              ^~~~
      |              |
      |              const char*
In file included from /tmp/arduino_modified_sketch_653462/TelnetStreamEsp32Test.ino:1:
/home/chepecarlos/5.Programas/3.Desarrollo/1.Arduino/arduino-1.8.19/libraries/WiFi/src/WiFi.h:79:21: note:   initializing argument 1 of 'int WiFiClass::begin(char*, const char*)'
   79 |     int begin(char* ssid, const char *passphrase);
      |               ~~~~~~^~~~
TelnetStreamEsp32Test:19:12: error: 'class WiFiClass' has no member named 'waitForConnectResult'
   19 |   if (WiFi.waitForConnectResult() != WL_CONNECTED) {
      |            ^~~~~~~~~~~~~~~~~~~~
exit status 1
invalid conversion from 'const char*' to 'char*' [-fpermissive]

Este informe podría contener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

---

svdrummer, 2023-01-21:
Looks like you are using ESP32 syntax in some areas.
I have compiled in ESP32 and 8266 many times, including today. The fault is not in the library

JAndrassy, 2023-01-21:
I made a commit with a fix

closed 2023-01-21 by JAndrassy

chepecarlos, 2023-01-23:
Hello, thanks for answer, but the problem is with the ESP8266.
Using Arduino ide 1.8.19 and addons esp8266 3.1.1

Attached image
[image]
Thanks for the support, I want to make a video of this library that seems great to me, that's why I already tried with esp32 and it works for me, but not with esp8266.

svdrummer, 2023-01-23:
I use this on esp32 and esp8266. what is the code snippet for your telnet print? Telnet.print Telnet.write?
The error reads as if you are doing something differently

chepecarlos, 2023-01-24:
I already understood the one that is in arduino ide is not updated, thanks for the support