JAndrassy/TelnetStream #27: Not compatible with esp8266/3.1.1

Roos-AID, 2023-01-28:
Cannot compile with esp8266 version 3.1.1 
Multiple errors

error: type 'Print' is not a base type for type 'ArduinoWiFiServer'
  110 |   using Print::write;

warning: 'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]
   42 |   WiFiClient accept() {return WiFiServer::available();}

JAndrassy, 2023-01-28:
update the library to 1.2.3

ccrawford, 2023-01-28:
Seeing same issue. PlatformIO says I'm using version=1.2.3. 
Here are some hints:

I'm using PlatformIO and and wemos d1 mini 8266. 

Just started working on some old code that was compiling a month ago with version 1.2.2. Dropping back to 1.2.2 still results in same errors, so something else changed.

I tried bringing the esp8266 package library down to 4.0.1 to test (Current is 4.1.0) but then was getting other errors compiling the framework, so bailing on that.

Roos-AID, 2023-01-28:
I am running 1.2.3 for sure, problem remains. When I revert to esp8266 3.0.2 it is ok.  Problem starts with 3.1.0

JAndrassy, 2023-01-28:
the problem is fixed in version 1.2.3. I tested it in Arduino IDE and Eclipse Sloeber (Arduino plugin)

ccrawford, 2023-01-28:
Compiles in the Arduino IDE for me, but not in PlatformIO (where I sure thought it was working a while ago.) Assume the difference is in the versions of C++ used (but I think they're both 11), or in the compiler options. But I'm no expert here. 

Platform IO is erroring because the using Print::write() statement and the write function overrides class ArduinoWifiServer does not derive from Print. I don't know why that isn't a problem in the Arduino IDE, but it clearly is not. Not sure if the use is really intended to be a namespace helper. But that wouldn't explain the override.

Roos-AID, 2023-01-28:
Arduino IDE 2.0.3 same problem here.  Will try to analyse later, but I am certain I have .1.2.3 included

ccrawford, 2023-01-28:
If I edit ArduinoWiFiServer.h to:
1. Delete line 110 ("using Print::write;")
2. Remove the override modifiers from line 62 and line 66 (the write functions)

then it compiles and the sample program runs in both Arduino IDE and PIO. I have *not* run extensive testing, just smoke tested it.

JAndrassy, 2023-01-29:
as workaround just delete ArduinoWiFiServer.h

Roos-AID, 2023-01-29:
Thanks,  this seems to fix it.  Telnet is running without problems.  But maybe somewhere else a problem in 3.1.1 as my app crashed after a while. Need to dig into that.

Laxilef, 2023-02-03:
The problem is still there.
Arduino IDE 2.0.3 / VS Code + Arduino plugin
ESP8266
TelnetStream 1.2.3

Removing the ArduinoWiFiServer.h file solved the problem. But this is not a production solution.

JAndrassy, 2023-02-03:
in sketch always include the networking library (ESP8266WiFi in this case) before TelnetStream.h

Laxilef, 2023-02-03:
Nope.
https://prnt.sc/ViNNJgrssELY

JAndrassy, 2023-02-03:
please try the new version 1.2.4

closed 2023-02-03 by JAndrassy

Roos-AID, 2023-02-04:
Thanks, this works. Both with ESP32 and with ESP8266.