JAndrassy/TelnetStream #30: Arduino OPTA

MauroKTM, 2023-07-20:
it returns error when compile for Arduino OPTA

`/Users/xxx/Documents/Arduino/libraries/TelnetStream/src/TelnetStream.cpp: In member function 'boolean TelnetStreamClass::disconnected()':
/Users/xxx/Documents/Arduino/libraries/TelnetStream/src/TelnetStream.cpp:27:7: error: no match for 'operator!' (operand type is 'arduino::EthernetServer')
   if (!server)
       ^~~~~~~
/Users/xxx/Documents/Arduino/libraries/TelnetStream/src/TelnetStream.cpp:27:7: note: candidate: operator!(bool) <built-in>
/Users/xxx/Documents/Arduino/libraries/TelnetStream/src/TelnetStream.cpp:27:7: note:   no known conversion for argument 1 from 'arduino::EthernetServer' to 'bool'

exit status 1

Compilation error: exit status 1`

JAndrassy, 2023-07-20:
modify the `#if` in `boolean TelnetStreamClass::disconnected() `

to
```
#if __has_include(<WiFiNINA.h>) || __has_include(<WiFi101.h>) || __has_include(<PortentaEthernet.h>)
```

MauroKTM, 2023-07-20:
It works.

thank you!

closed 2023-07-20 by MauroKTM

MauroKTM, 2023-07-21:
It compile now, but it seems not start any telnet server.
Any suggestion?