JAndrassy/TelnetStream #32: TelnetPrint.accept() for WiFiS3.h

sebaJoSt, 2023-09-01:
Hey, thanks for you work and updating to Arduino's WiFiS3 library.

I just tried to do some tests with the new Arduino Uno R4 WiFi but unfortunately I can't get my advanced chat server example to work because:
```
 error: 'class WiFiServer' has no member named 'accept'
```
Is it possible to implement this feature? 

Here's the full example:
```
#include <WiFiS3.h>
#include <TelnetPrint.h>
#include "arduino_secrets.h"
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;  // your network SSID (name)
char pass[] = SECRET_PASS;  // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;           // your network key index number (needed only for WEP)
int status = WL_IDLE_STATUS;
boolean alreadyConnected = false;  // whether or not the client was connected previously

NetClient *Clients;

#define MAX_CLIENTS 8
#define SERVER_PORT 23

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ;  // wait for serial port to connect. Needed for native USB port only
  }

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true)
      ;
  }

  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to SSID: ");
    Serial.println(ssid);
    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  Clients = new NetClient[MAX_CLIENTS];
  // start the server:
  TelnetPrint = NetServer(23);
  TelnetPrint.begin();

  // you're connected now, so print out the status:
  printWifiStatus();
}

void loop() {

  //----Advanced Chat Server--------------
  NetClient newClient = TelnetPrint.accept();

  if (newClient) {

    for (byte i = 0; i < MAX_CLIENTS; i++) {
      if (!Clients[i]) {
        Serial.print("We have a new client #");
        Serial.println(i);
        newClient.print("Hello, client number: ");
        newClient.println(i);
        // Once we "accept", the client is no longer tracked by the server
        // so we must store it into our list of clients
        Clients[i] = newClient;

        break;
      }
    }
  }

  for (byte i = 0; i < MAX_CLIENTS; i++) {
    if (Clients[i] && Clients[i].connected()) {
      if (Clients[i].available()) {
        while (Clients[i].available()) {
          char rc = Clients[i].read();
          Serial.print(rc);
        }
      }
    }
  }

  // stop any clients which disconnect
  for (byte i = 0; i < MAX_CLIENTS; i++) {
    if (Clients[i] && !Clients[i].connected()) {
      Clients[i].stop();
      Serial.print("Client #");
      Serial.print(i);
      Serial.println(" disconnected");
    }
  }
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
}

```

closed 2023-09-01 by JAndrassy

sebaJoSt, 2023-09-01:
Understand..I hoped it could be done similar to esp32/esp8266. So i need to wait until accept() is supported by WifiS3. Thanks for clarifying this.

JAndrassy, 2023-09-02:
other WiFi libraries by Arduino don't have accept()

sebaJoSt, 2023-09-02:
I use TelnetStream as a dependency in my own library [BlaeckTCP](https://github.com/sebaJoSt/BlaeckTCP) to enable WIFI and Ethernet functionality. So if there's no support for accept(), I think I try to implement Simple Chat Server as a fallback solution for Arduino's WiFi libs. I'm quite new to this WiFi/Ethernet libs and thought there's more feature parity across architectures.

JAndrassy, 2023-09-02:
accept() is in Arduino Ethernet, new versions of esp8266 Arduino (as alias to available()) and in my libraries EthetnetENC and WiFiEspAT. 
WiFiNINA and WiFi101 have my PR adding accept() waiting long time now.

sebaJoSt, 2023-09-02:
It's a pity those PRs are not already pulled in, yet.  I stumbled upon the names of your libs before, but didn't know they were from you. I really appreciate your work for the Arduino community!