JAndrassy/TelnetStream #34: TelnetStream with RP2040 and W5100s not working

ZinnerC, 2023-10-20:
I would like to use TelnetStream to have an option to have a telnet connection for remote debugging. However, it does not work with [Arduino Core](https://github.com/earlephilhower/arduino-pico). Any chance that it will get an update?

JAndrassy, 2023-10-20:
you attempt to use it with the  lwip_5100 library? that is not supported.

> TelnetStream/TelnetPrint works as it is with esp8266 and esp32 WiFi library, with the Ethernet and EthernetENC, with WiFiNINA, WiFi101, WiFiS3 and WiFiEspAT library.

ZinnerC, 2023-10-20:
Yes, I tried it to use with the lwip_5100 library. Is there an alternative you would suggest?

JAndrassy, 2023-10-20:
> TelnetStream/TelnetPrint works as it is with esp8266 and esp32 WiFi library, with the **Ethernet** and EthernetENC, with WiFiNINA, WiFi101, WiFiS3 and WiFiEspAT library.

ZinnerC, 2023-10-20:
Ok thanks. Well, I need to stay with lwip_5100 due to OTA on the rp2040. Lets see. Thanks still.

closed 2023-10-20 by ZinnerC

JAndrassy, 2023-10-20:
so I will try to make TelnetStream work with the lwip libraries.
but I have OTA for RP2040 with the Ethernet library. 
https://github.com/JAndrassy/ArduinoOTA#arduino-library-to-update-the-sketch-over-the-network

JAndrassy, 2023-10-20:
I made the necessary changes. now TelnetStream supports the networking libraries of the Pico Core

ZinnerC, 2023-10-22:
Wow great. Works like a charm.