JAndrassy/TelnetStream #35: TelnetStream.flush(); does nothing on esp32

xgnata, 2023-11-11:
In ArduinoWiFiServerESP.h we read:
#ifndef ESP32 // has wrong client.flush()
    for (uint8_t i = 0; i < MAX_MONITORED_CLIENTS; i++) {
      if (connectedClients[i]) {
        connectedClients[i].flush();
      }
    }
#endif

therefore flush does nothing on esp32?

Why is it so?
Which bug should be solved in esp32 libs to make TelnetStream.flush() work.

JAndrassy, 2023-11-12:
esp32 doesn't have true client.flush(). client.flush clears the receive buffer there.

svdrummer, 2023-11-12:
One of the code examples actually says no clear in the title. Have a look at that.

JAndrassy, 2023-11-12:
https://github.com/espressif/arduino-esp32/pull/8871

xgnata, 2023-11-12:
Thanks for the answers!
I understand that the wifi flush is not a flush but a clear which throws away the buffer.
Ok, but is there a way to implement a real "flush : Output whatever is in the queue/buffer via the device or port" ?
I'm using TelnetStream to log/debug several xtasks. A mutex to protect a flush does the job nicely for the Serial port : no messages are mixed.
I'm going to implement a short delay to make sure, as much as possible, that the wifi buffer is flushed....but this is ugly. Anyway better way? Check if the buffer is empty? how?

JAndrassy, 2023-11-12:
I don't know

closed 2024-02-05 by JAndrassy