JAndrassy/TelnetStream #37: NetApiHelpers flush override error

Vildan, 2023-12-21:
Error when compiling 1.3.0 version. 
.pio/libdeps/esp32dev/NetApiHelpers/src/ServerTemplate.h:118:16: error: 'void ServerTemplate<TServer, TClient>::flush() [with TServer = WiFiServer; TClient = WiFiClient]' marked 'override', but does not override
it was not there on 1.2.4 version.

Vildan, 2023-12-21:
ESP32 board

JAndrassy, 2023-12-21:
what version of ESP32 core?

Vildan, 2023-12-21:
[env:cube1]
platform = espressif32
board = esp32dev
framework = arduino
build_flags = -DHARDWARE_CUBE1
			  -D PIO_ENV_NAME=\"${PIOENV}\"
board_build.partitions = default.csv
monitor_speed = 115200
extra_scripts = post_build_script.py
lib_deps = 
	...
	jandrassy/TelnetStream@^1.3.0

JAndrassy, 2023-12-21:
what version of the esp32 core do you have installed?
only old versions before 2.0.3 from May 2022 don't have flush() in Print

Vildan, 2023-12-22:
After updating platform Espressif 32 from 3.1.0 version to 6.4.0 version, it start compiling without the problem. Thank you

JAndrassy, 2023-12-22:
that is the platformio versions numbering, not the normal one

closed 2023-12-22 by JAndrassy