JAndrassy/TelnetStream #39: 'class EthernetServer' has no member named 'flush'

mduino, 2024-03-11:
Hi, I'm having compilation errors with the library.
I'm working on an Arduino Mega with an Ethernet shield that works with the standard Ethernet library.

However this is the errors I get when I try uploading the TelnetPrintEthTest example:

```
c:\Users\...\Documents\Arduino\libraries\TelnetStream\src\TelnetStream.cpp: In member function 'virtual void TelnetStreamClass::flush()':
c:\Users\...\Documents\Arduino\libraries\TelnetStream\src\TelnetStream.cpp:66:10: error: 'class EthernetServer' has no member named 'flush' server.flush();
```

mduino, 2024-03-11:
Okay, I think I was able to track down the problem:
I'm using an Arduino Mega that is build in an Industrial Shield module that has it's own Board Manager. I suspect that the library is not able to properly detect my board.
I'll investigate further and share my findings.

JAndrassy, 2024-03-11:
it has outdated Arduino Core. Print.flush is missing

mduino, 2024-03-11:
It was indeed a problem with the Board Manager. The newest release of their custom board manager was missing these functions.
Thank you!

closed 2024-03-11 by mduino