JAndrassy/TelnetStream #42: Error when compiling with ESP32 3.0.0

Spangen, 2024-05-29:
Just a heads up, flush() has been removed in ESP32 3.0.0. 

[https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html](url)

Compiling telnetstream.h fails with:

`In file included from c:\Arduino\Code\libraries\NetApiHelpers\src/ArduinoWiFiServer.h:48,
                 from c:\Arduino\Code\libraries\TelnetStream\src/NetTypes.h:38,
                 from c:\Arduino\Code\libraries\TelnetStream\src/TelnetStream.h:22,
                 from C:\Arduino\Code\Human-Precense\LD2410\SensorSetupOTA_MultiWiFi\OTA.h:12,
                 from C:\Arduino\Code\Human-Precense\LD2410\SensorSetupOTA_MultiWiFi\SensorSetupOTA_MultiWiFi.ino:6:
c:\Arduino\Code\libraries\NetApiHelpers\src/ServerTemplate.h: In instantiation of 'class ServerTemplate<NetworkServer, NetworkClient>':
c:\Users\Robert\Dropbox\Cisco\ESP8266 ESP32\Arduino\Code\libraries\TelnetStream\src/TelnetStream.h:27:13:   required from here
c:\Arduino\Code\libraries\NetApiHelpers\src/ServerTemplate.h:116:16: error: type 'Print' is not a base type for type 'ServerTemplate<NetworkServer, NetworkClient>'
  116 |   using Print::write;
      |                ^~~~~
c:\Arduino\Code\libraries\NetApiHelpers\src/ServerTemplate.h:70:18: error: 'size_t ServerTemplate<TServer, TClient>::write(uint8_t) [with TServer = NetworkServer; TClient = NetworkClient; size_t = unsigned int; uint8_t = unsigned char]' marked 'override', but does not override
   70 |   virtual size_t write(uint8_t b) override {
      |                  ^~~~~
c:\Arduino\Code\libraries\NetApiHelpers\src/ServerTemplate.h:74:18: error: 'size_t ServerTemplate<TServer, TClient>::write(const uint8_t*, size_t) [with TServer = NetworkServer; TClient = NetworkClient; size_t = unsigned int; uint8_t = unsigned char]' marked 'override', but does not override
   74 |   virtual size_t write(const uint8_t *buf, size_t size) override {
      |                  ^~~~~
c:\Arduino\Code\libraries\NetApiHelpers\src/ServerTemplate.h:118:16: error: 'void ServerTemplate<TServer, TClient>::flush() [with TServer = NetworkServer; TClient = NetworkClient]' marked 'override', but does not override
  118 |   virtual void flush() override {
      |                ^~~~~

exit status 1

Compilation error: exit status 1`

73 de \\Spangen

JAndrassy, 2024-05-29:
I do a release for NetAPiHelpers library later today.
you can use the version from GitHub if you don't want to wait.
https://github.com/Networking-for-Arduino/NetApiHelpers

JAndrassy, 2024-05-29:
NetApiHelpers 1.0.2 is in Library Manager

closed 2024-05-29 by JAndrassy

JAndrassy, 2024-05-29:
btw: I made that change in esp32 platform, which required a change in NetApiHelpers