JAndrassy/TelnetStream #43: Espressif ESP32 3.0.0 not handling Print capability?

thorathome, 2024-05-29:
Hi. Thanks.  

I'm using the Arduino IDE most recent version on Win 11 and just moved to Espressif's ESP32 3.0.0 Board package from 2.0.17.  

Getting new error messages not seen with previous Espressif ESP32 2.0.x versions. Been using your TelnetStream library for years. Love it!

Any thoughts?  
Thanks again. 
```
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\NetApiHelpers\src/ServerTemplate.h: In instantiation of 'class ServerTemplate<NetworkServer, NetworkClient>':
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\TelnetStream\src/TelnetStream.h:27:13:   required from here
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\NetApiHelpers\src/ServerTemplate.h:116:16: error: type 'Print' is not a base type for type 'ServerTemplate<NetworkServer, NetworkClient>'
  116 |   using Print::write;
      |                ^~~~~
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\NetApiHelpers\src/ServerTemplate.h:70:18: error: 'size_t ServerTemplate<TServer, TClient>::write(uint8_t) [with TServer = NetworkServer; TClient = NetworkClient; size_t = unsigned int; uint8_t = unsigned char]' marked 'override', but does not override
   70 |   virtual size_t write(uint8_t b) override {
      |                  ^~~~~
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\NetApiHelpers\src/ServerTemplate.h:74:18: error: 'size_t ServerTemplate<TServer, TClient>::write(const uint8_t*, size_t) [with TServer = NetworkServer; TClient = NetworkClient; size_t = unsigned int; uint8_t = unsigned char]' marked 'override', but does not override
   74 |   virtual size_t write(const uint8_t *buf, size_t size) override {
      |                  ^~~~~
c:\Users\thor\Dropbox\My Documents\Arduino\libraries\NetApiHelpers\src/ServerTemplate.h:118:16: error: 'void ServerTemplate<TServer, TClient>::flush() [with TServer = NetworkServer; TClient = NetworkClient]' marked 'override', but does not override
  118 |   virtual void flush() override {
```

JAndrassy, 2024-05-29:
https://github.com/JAndrassy/TelnetStream/issues/42

closed 2024-05-29 by JAndrassy

thorathome, 2024-05-29:
YOU are unbelievably responsive! Thank you for the immediate fix. 

 

Compiles working again.  TelnetStream working again. I am amazed.  

 

Much appreciated and well done.  

 

Closed #43 <https://github.com/JAndrassy/TelnetStream/issues/43>  as completed.